SpenceKonde/DxCore #557: Should Multiple PWM Outputs Function Concurrently?

microPaul, 2024-12-08:
I seem to continue to be PWM challenged.

I'm using a homebrew board with an AVR64DB64.  From the documentation on the DxCore GitHub page it looks like TCA0 will provide six PWM outputs.  And I do get six PWM outputs on Port D (I change the value of PORTMUX.TCAROUTEA to 0x03).  My "problem" is that I only can get one of the PWM outputs to work at a time.  If I have multiple AnalogWrite statements, like:
```    analogWrite(PIN_PD0, 64);
    analogWrite(PIN_PD1, 32);
    analogWrite(PIN_PD2, 128);
    analogWrite(PIN_PD3, 190);  
    analogWrite(PIN_PD4, 45);     
    analogWrite(PIN_PD5, 99); 
  ```

then only the last statement is effective.  Is this the way PWM is supposed to work?  Only one TCA0 PWM output active at a time?   I thought (or was hoping) that I could get six concurrently operating PWM outputs from one TCA timer.  Please let me know if the one-at-a-time PWM output that I'm seeing is the intended functionality.   I'm using the 1.5.11 version of the DxCore.

Thanks!

microPaul, 2024-12-08:
Here's the whole program that I'm using for test purposes:

```
#undef Serial
#define Serial Serial4  // Using UART4 No Alt (PE0, PE1)
                                   
void setup() {

    //Serial.swap(0); // no need to swap for this UART pin assignment
    Serial.begin(57600, SERIAL_8N1); 
    Serial.println("");   
    Serial.println("_test_pwm_TCA_DB64_241207-01"); 
    Serial.println("");  
    delay(500); // give serial port time to send data

    PORTA.OUTCLR = PIN7_bm;  // PA7 - LED off
    PORTA.DIRSET = PIN7_bm;  // LED DDR set for output
    
    // now set up things for PWM output.  Trying to get six concurrent PWM outputs from TCA0 on Port D
    Serial.printf("PORTMUX.TCAROUTEA = 0x%02x\n", PORTMUX.TCAROUTEA);
    PORTMUX.TCAROUTEA = 0x03; // select D output
    
    analogWrite(PIN_PD0, 64);
    analogWrite(PIN_PD1, 32);
    analogWrite(PIN_PD2, 128);
    analogWrite(PIN_PD3, 190);  
    analogWrite(PIN_PD4, 45);     
    analogWrite(PIN_PD5, 99);      
    Serial.printf("PORTMUX.TCAROUTEA = 0x%02x\n", PORTMUX.TCAROUTEA);
}

////////////////////////////////////////////
//
void loop() {
  // loop is not a loop, it's a function that is repeatedly called
  
  uint32_t timeRef = millis();  // establish a time reference variable

  
  while (1) {
    // now we have a loop
   
    if (millis() - timeRef > 1000) { // see if timeout
      timeRef = millis();  re-init the timer referrence

      led0Toggle();
    } 
    
  }
  
}

void led0On() {
  PORTA.OUTSET = PIN7_bm;  // Port Pin Set High
}

void led0Off() {
  PORTA.OUTCLR = PIN7_bm;  // Port Pin Set High
}

void led0Toggle() {
  PORTA.OUTTGL = PIN7_bm; // toggle Port Pin
} 
```

SpenceKonde, 2025-05-25:
I am greatly concerned upon reading this, it is a high priority for investigation.