SpenceKonde/DxCore #559: New Curiosity Nano boards / avrdude

aikopras, 2025-01-11:
To test if my own libraries run on the various DxCore processors, I bought a couple of Curiosity nano's.

First I tested with the AVR128DA48, using the Curiosity Nano programmer, and everything went fine.

Now I want to test the curiosity boards for the AVR64EA48 (and AVR64DD32), but run into problems.

1) avrdude complaints that it doesn't know the EA chips. I understood this problem comes from the outdated (6.3.0 ) avrdude version that is shipped with DxCore. Since I already had avrdude V8 on my system, I used their new executable and config files to replace the old in my (MACOS) Library/Arduino15/packages/DxCore/tools/avrdude/6.3.0-arduino17or18/bin and /etc folders. 

2) After that was done, the Arduino IDE had problems with the avrdude config file in Library/Arduino15/packages/DxCore/hardware/megaavr/1.5.11 . So I replaced this file with the new V8 config file that was in the /etc folder. 

3) Now I received the error "cannot find programmer id curiosity_updi". I understood that some lines were missing from the configuration file. See https://github.com/platformio/platform-atmelmegaavr/issues/31 

4) I added the missing lines in the config file, but now I receive the message "Error: programmer curiosity_updi and part AVR64EA48 have no programming modes in common".

Did I do the right steps above? If yes, how should I continue to get avrdude uploading to the EA Curiosity boards?

aikopras, 2025-01-15:
At the end I decided to compile using DxCore, and upload using Atmel Start (on my MAC onder VirtualBox in a W10 environment). That works well.

It would be nice if, on the longer term, DxCore would also be able to directly Flash the EA and DD curiosity boards. Let me conclude by saying how impressed and grateful I am for the fantastic work that has / is done to develop DxCore (and megaTinyCore). Thanks!!!

mcuee, 2025-02-25:
What if you export the hex file and program using avrdude 8.0?

aikopras, 2025-02-25:
I assume that would work as well. My current workaround is already the creation of a HEX file. But instead of avrdude 8.0, I used Atmel Start

MCUdude, 2025-06-10:
> I added the missing lines in the config file, but now I receive the message "Error: programmer curiosity_updi and part AVR64EA48 have no programming modes in common".

If you use Avrdude 8.0, replace `curiosity_updi` with `pkobn_updi` in programmers.txt.

The reason for the error is that the link you referred to was from 2021, before Avrdude added a new field that tells Avrdude which interfaces (ISP, JTAG, UPDI, etc.) the programmer supports.