SpenceKonde/DxCore #562: Spence, are you doing OK?

microPaul, 2025-02-27:
I've been getting more and more concerned about Spence's personal well being.  I hope he's OK and just too busy with day-job stuff to post here.  I don't think he's posted on his Git Hub space for several months.  

Is anyone here personal friends with Spence and could let us know how he's doing?

CreativeRobotics, 2025-02-28:
I posted a bug report in early November and he replied mid December ... so maybe just very busy?

mikrocoder, 2025-05-12:
Spence does not respond to any request. The offers on the subject of pinout diagrams remained completely unanswered. Something is not right.

CreativeRobotics, 2025-05-12:
Spence has been active on MegaTinyCore in the last few days.

mikrocoder, 2025-05-13:
Why doesn't he respond to the offers for his pinout requests? There are enough posts on this. Not a single one has been answered. I wanted to help with that too. But I no longer feel motivated.

MCUdude, 2025-05-13:
Well, life has its seasons. As an Arduino core maintainer myself for almost 10 years, I can honestly say that there are times when the interest in maintaining these cores isn't all that high. And it really sucks when the motivation is low and users are demanding urgent changes, new features or bugfixes, simply because their project relies on your core.

Maybe @SpenceKonde feels the same way? He may be busy with other things (the README says something about tax issues), or maybe just wants to spend his time on different things. He probably has his reasons, and that's perfectly reasonable.

A statement from Spence with a quick, few quick words on what we can expect in the future would, of course, be welcome. But I think we should be humble and give him some slack.

If you're worried or have other questions, you can always send him an email.

mikrocoder, 2025-05-14:
All correct. The point is, he could have at least left a comment in the pinout discussion threads. I just expect that. Several people have thought to help him with the missing pinouts. There was no response. If he doesn't have time, he can say so. I expect that.

closed 2025-06-06 by SpenceKonde