SpenceKonde/DxCore #566: analogRead()

domBavay, 2025-05-19:
hello,
First of all, thanks for that great job :)
I am using AVR64DD20 on arduino ide with DxCore.
I have a problem with analogRead. at compilation, i get that message:
"analogRead called with constant pin that is not a valid analog pin" with: analogRead(PIN_PC3)
i don't really know what that means :/ i made a previous project on mplabX with the same mcu, and i could use pin PC3 as ADC.
What could be wrong in my project? maybe i have to configure PIN_
PC3 as an inpu? (i have the classic pinMode command) but is it enough?
Thanks :)
dominique

DrItanium, 2025-05-24:
I just noticed that if MVIO is enabled then analog pins on that port are disabled. Perhaps that is why

SpenceKonde, 2025-06-06:
Yes. Note also the open issue relating to problems disabling MVIO

domBavay, 2025-06-06:
Hello, yes that was the problem, i just read about it this morning, and was about to write it on the forum :)
Thanks for all :)
dominique