SpenceKonde/DxCore #571: EEPROM not working on EA devices

CreativeRobotics, 2025-07-30:
I can't get the EEPROM to work on my 64EA28.

I tried first with sequential single byte write then read and all I get on a read is 255. I just tried with the code below using structs and the put() and get() methods but I am also gettingnothing back byt 255. Any idea what is going on here?

```
#include <EEPROM.h>

typedef struct dta_t{
  uint8_t dta1;
  uint8_t dta2;
} dta_t;

uint8_t n = 0;
uint16_t addr = 0;

dta_t data;

//eeAddress += sizeof(float); // Move address to the next byte after float 'f'.

//EEPROM.put(eeAddress, customVar);
void setup() {
  Serial.begin(115200);
  data.dta1 = 12;
  data.dta2 = 16;
}

void loop() {
  EEPROM.put(addr, data);
  delay(1);
  Serial.print("Wrote ");
  Serial.print(data.dta1);
  Serial.print(", ");
  Serial.print(data.dta1);
  dta_t newData;
  EEPROM.get(addr, newData);
  Serial.print(" Read Back ");
  Serial.print(newData.dta1);
  Serial.print(", ");
  Serial.println(newData.dta1);
  
  data.dta1++;
  data.dta2++;
  delay(100);
}
```

SpenceKonde, 2025-07-30:
No idea, I'll put it on the list to look at once I get compile fixed on my development version and get the upload problem sorted out with serialupdi, since I can't even upload to them right now.

CreativeRobotics, 2025-07-31:
Thanks - Uploading would be nice, I currently use 'export compiled binary' and have Microchip Studio programmer opened and pointing at the HEX file that Arduino generates. Works well enough.

I was trying to cobble something together from my limited understanding of the datasheet and the EEPROM library. Shockingly, the code below appears to work - I put this function in my code and now I can write a load of bytes to different addresses, then read them back using EEPROM.read() and they are still there after the device has been powered off and on again ...

```
void writeEEVal(uint16_t index, uint8_t in){
  uint8_t oldSREG = SREG;
  while (NVMCTRL.STATUS & NVMCTRL_EEBUSY_bm);
  // check at start - not end. Makes writing single bytes fasterms
  // Note that we only have interrupts disabled for the dozen or so clock cycles
  // during which we *start* the write, not for the while loop, and we save SREG
  // before the while loop. That way there is only a 1 clock window where an
  // interrupt that starts a write will cause this write to halt the CPU
  // which would cause millis() to lose time.
  // Note that writing more than 1 byte in an ISR will **always** cause millis to lose time.
  cli();
  _PROTECTED_WRITE_SPM(NVMCTRL.CTRLA, NVMCTRL_CMD_EEPERW_gc);
  *(uint8_t *)(MAPPED_EEPROM_START + (index & EEPROM_INDEX_MASK)) = in;

  SREG = oldSREG; // restore SREG and interrupts
}
```

commit 5db0aace (2025-08-14): Fix #571 Hopefully get tests compiling
closed 2025-08-14 by SpenceKonde