SpenceKonde/DxCore #574: Multiple tests are broken

SpenceKonde, 2025-08-14:
Anyone know why these have failed?

https://github.com/SpenceKonde/DxCore/actions/runs/16974761236

https://github.com/SpenceKonde/DxCore/actions/runs/16974761237

nomakewan, 2025-08-17:
This is happening because the `version` value in `platform.txt` does not follow standards:

https://github.com/SpenceKonde/DxCore/blob/eb778d364b4e087f5ae8a30456d627a3e9194a0b/megaavr/platform.txt#L19

The `dev` part is invalid in the semantic versioning spec. SEE: https://regex101.com/r/Ly7O1x/3/

SpenceKonde, 2025-10-22:
Action not unsolved - PR's to fix it were submitted and are sitting unmerged. 

Is there a way to fork a github action and put the fix in? Can you just fork an action and change the reference to your fork? Does that work?
Or is there some other hocus pocus you need to do? I'm

closed 2026-04-18 by SpenceKonde